spacelab-ufsc/obdh2 #30: Hardware: Develop the circuit of the MT25QL01GBBB memory IC

mgm8, 2019-12-03:


mgm8, 2019-12-05:
From the [datasheet](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/mt25q/die-rev-b/mt25q_qlkt_l_01g_bbb_0.pdf) (page 79):

*"Normal precautions must be taken for supply line decoupling to stabilize the VCC supply. Each device in a system should have the VCC line decoupled by a suitable capacitor (**typically 100nF**) close to the package pins. At power-down, when VCC drops from the operating voltage to below the power-on-reset threshold voltage shown here, all operations are disabled and the device does not respond to any command."*

mgm8, 2019-12-05:
Reference design: [Technical Note - PCB Layout Design Guidelines](https://www.micron.com/-/media/client/global/documents/products/technical-note/nor-flash/tn2509_layout_guidelines_spi_nor_mt25q.pdf)

andrempmattos, 2019-12-11:
MT25QL01GBBB8ESF-0SIT

andrempmattos, 2019-12-12:
65a95e1f6fb60cf8cc9d04b45ac07585bf1c6b48

closed 2020-01-11 by andrempmattos