spacelab-ufsc/obdh2 #95: Architecture: Define a memory management system

andrempmattos, 2020-01-28:
**Problem:** Since OBDH receive and request data from the other modules and peripherals with different rates and and sizes, the non volatile memory should have a management system coordinating the store and read routines besides the storage of critical data and a method to handle multual access of the memory, a shared resource. In the previous architecture, the design was settle in queues to receive data from other tasks to an unified memory task, which it was the unique task allowed to perform memory operations. This approach caused issues in case of hardware problems due to unprotected queues and misuse of the operating system.

**Solution 1:** Continue using the previous archtecture. However, using properly methods to handle the queues and more effort to avoid deadlocks in case of failures.

**Solution 2:** Fragmented permissions and access to the memory operations. This method should consider that the input data are received in different rates and sizes, the data stored need to be retrieved in packets with the same timestamp, data race hazards and interleaved accesses to the memory.

**Solution 3:** Usage of a light weight library file system or memory management algorithms.

mgm8, 2020-04-15:
Possible solution: A file system like [FatFS](http://elm-chan.org/fsw/ff/00index_e.html) or [SafeFLASH](https://www.hcc-embedded.com/safeflash).

mgm8, 2020-04-15:
[https://www.hitex.com/fileadmin/documents/sw_components/AN101_choosing_a_file_system_hitex.pdf](https://www.hitex.com/fileadmin/documents/sw_components/AN101_choosing_a_file_system_hitex.pdf)

mgm8, 2020-04-24:
Another possible solution: [Reliance Edge](https://github.com/datalightinc/reliance-edge)

mgm8, 2020-04-24:
All three file systems will be tested. All the satellite data will be stored in files.

closed 2020-04-24 by mgm8