spacelab-ufsc/obdh2 #329: The OBDH2 cannot communicate with the EPS2.

LucasAzulay, 2023-03-09:
The OBDH2 cannot communicate with the EPS2.

LucasAzulay, 2023-03-16:
Thanks for the response, we implemented the EPS 2.0 code in the dev_firmware version and the communication between the OBDH 2.0 and EPS 2.0 worked as intended, the EPS recognized the read requests (cmd = 1), but we were not able to receive the data in the OBDH side (highlighted in the image).

LucasAzulay, 2023-03-16:
Thanks for the response, we implemented the EPS 2.0 code in the dev_firmware version and the communication between the OBDH 2.0 and EPS 2.0 worked as intended, the EPS recognized the read requests (cmd = 1), but we were not able to receive the data in the OBDH side (highlighted in the image).
[image]

joaoclaudioeb, 2023-03-17:
@LucasAzulay, could you tell me if the TTC 2.0 was connected with OBDH 2.0 and EPS 2.0 during these tests that you conducted?

LucasAzulay, 2023-03-17:
Yes, the three were connected.

joaoclaudioeb, 2023-03-17:
Ok. Today I will make some small changes to the OBDH's firmware in the _dev_firmware_. I will then contact you again to ask you to do some testing.

joaoclaudioeb, 2023-03-17:
@LucasAzulay, could you please use the code in _wdt_bug_fix_ into the OBDH 2.0? You should see EPS 2.0 being able to receive multiple commands from OBDH 2.0. OBDH 2.0 should not show any EPS 2.0 related errors. Finally, you will still see TTC 2.0 related errors: 
[image]

LucasAzulay, 2023-03-22:
We tested using the dev_firmware into the OBDH and we still have this erros.

[image]

joaoclaudioeb, 2023-03-22:
Could you please tell me if these errors occur recurrently? Here we have not yet observed these errors.

LucasAzulay, 2023-03-29:
These errors still occurring. The OBDH is not receiving data from the EPS (image 2), and the EPS is receiving the commands correctly most of the time, but sometimes the commands are getting corrupted (image 4).
We are using the TTC 2.0 v0.4.2, OBDH 2.0 v10.0 and EPS v0.3.0.

[image]
[image]
[image]
[image]

joaoclaudioeb, 2023-04-06:
@LucasAzulay, sorry for the delayed response. We have not yet been able to replicate the errors, however, I have some observations:

1) Please check if in config/config.h the _CONFIG_SET_DUMMY_EPS_ is set to 0 or 1. I suggest trying to do the tests with this deactivated, that is, with 0.

2) I would like to ask you to do a little "test". Would it be possible to change the position of the OBDH in your FlatSat platform? You can put it in any other slot. 

Please report me if something changed during the tests. We will try to make more tests to replicate the errors.