spacelab-ufsc/obdh2 #331: The OBDH2 cannot communicate with the Antenna

LucasAzulay, 2023-03-16:
We were testing the communication of the OBDH2 with the antenna. At first, the I2C communication between the OBDH2 and the antenna module fails. The OBDH2 I2C communication with the antenna is already completed? 
Also, the connections between the Antenna and the OBDH are correct? (images 2 and 3)
[image]
[image]
[image]

miguelboing, 2023-03-20:
The antenna implementation was imported from OBDH 1.0 and wasn't tested in OBDH 2.0. I will look into it to see if I find the problem. But for now, could you run the dev_firmware code and tell me if there is any difference in the logs?

miguelboing, 2023-03-22:
Hello Lucas, the problem was related to the isis_antenna driver, during the initialization. If you change [this line](https://github.com/spacelab-ufsc/obdh2/commit/3d4fadd2005a108838dac9ee30150c8e7811b87b) the I2C bus will be able to correctly initialize.

commit 3143e663 (2023-09-19): firmware: tests: drivers: Updating the ISIS Antenna driver test. closes #331
commit 57ba0545 (2023-09-19): Merge pull request #339 from spacelab-ufsc/aldebaran_mission

Merge pull request from aldebaran_mission, closes #334, fix #331